gawlikaj/elek-labirynt #11: Implement I2C driver for Qwiic Joystick

gawlikaj, 2021-04-15:
Read the data from the Qwiic Joystick for the x-axis, y-axis and select button on the joystick.  For the purposes of testing I will be using the UART debugger to output the data read from the Qwiic Joystick.

A strategy I have here is to do two things:
1) allow to read a single parameter from the joystick through individual functions (or one function with an enumeration specifying the value to be read)
2) allow to read all parameters through a single function call

This will be beneficial when I get to the game engine.  Or for that matter, investigating how I may look at building something more sophisticated later down the road.

gawlikaj, 2021-04-27:
Since this also is related to breadboard testing I have added this to the breadboard testing project as well...

gawlikaj, 2021-04-27:
The Joystick driver should not need to be incorporated into breadboard testing directly as the I2C communication for this is connected directly via the Qwiic cable between the ESP32 Thing Plus and the Joystick breakout board, however, when including other I2C devices (such as the Fuel Gauge) which may not have holes for standoffs, it may be necessary to use the pins instead of the Qwiic cable.  In the case of having another device on the I2C bus, it may be prudent to test on the breadboard to ensure that there are no conflicts.

gawlikaj, 2021-05-03:
Reviewing the example code for the I2C project to get an idea... also as identified in the pinout assignments spreadsheet:
SDA - GPIO23
SCL - GPIO22

These pins are essential for working with the Qwiic connect system.

gawlikaj, 2021-05-03:
The hookup guide provided on SparkFun provides the register information for which to retrieve the values of the button and axis:
https://learn.sparkfun.com/tutorials/qwiic-joystick-hookup-guide?_ga=2.252622153.1710335758.1620067209-1450422450.1611246352

The specific information of importance:
Address - 0x20
0x03      - Horizontal Position MSB 
0x04      - Horizontal Position LSB
0x05      - Vertical Position MSB
0x06      - Vertical Position LSB
0x07      - Current Button Position
0x08      - Button Status (Clears after last read)

gawlikaj, 2021-05-06:
starting work on the driver

gawlikaj, 2021-05-06:
I actually have split up the joystick into the qwiic component (for initializing the qwiic system) and the joystick driver itself (for interacting with the qwiic joystick)... this way I can instantiate the qwiic system independently even if I have other qwiic devices.

gawlikaj, 2021-05-07:
a test application has been prepared and the basic functions have been laid out.  Currently in the process of debugging/testing the joystick and qwiic drivers.  There does not seem to be any errors with the joystick portion, for now focusing on the qwiic portion with regards to initialization, transmitting data from the controller to the peripheral (joystick), receiving data from the peripheral (joystick). Any other qwiic I2C component should be easier to implement if connected on the same I2C bus,

gawlikaj, 2021-05-07:
In an effort to test the joystick first, I will try testing with the example code provided in the esp-idf (i2c_tools) to see it it works correctly.

gawlikaj, 2021-05-12:
modeling this after the arduino library functionality provided on SparkFun's website, but I have noticed that I am no longer getting the invalid state error..

the timeout error before was really  a failure for ack (the error message was miss-mapped).  So now the error messages are mapped correctly and the failure for ack was addressed by only using 7-bit addressing and making sure that the address was shifted by 1 and appending the bit for reading/writing as part of the I2C specification.  Now there really is a timeout error and am looking into it (there are also no more invalid state errors).  Also the sections for reading and writing have been broken apart in the qwiic_controller_read_from_peripheral and qwiic_controller_write_to_peripheral.

qwiic_controller_read_from_peripheral has two separate commands (as defined in the I2C driver library from the ESP-IDF)
writing the address of the peripheral and the register to query to the I2C bus
writing the address of the peripheral to the I2C bus and reading the response from the I2C bus

qwiic_controller_write_to_peripheral has a single command (as defined in the I2C driver library from the ESP-IDF)
writing the address of the peripheral, the register to be written to and the data being written

gawlikaj, 2021-05-12:
Currently when reading the data from the joystick I get the following values:
vertical = 16386
horizontal = 16386
select status = 0

the vertical and horizontal values are outside of the range possible from the 10-bit ADC (0-1023)
the select status value reflects not pushing down a button, however, even when the button is pressed it will not change the value

the commands are acknowledged correctly since all command parts have an acknowledge and are not refused.  So the Thing Plus "sees" the joystick.... looking into adding a wait for the response.

Also since each call to qwiic_joystick_chk_status always writes to the select status register after reading from it, I know that writing is working correctly as the command returns with ESP_OK.

gawlikaj, 2021-05-12:
As I thought about it when I was working on the UART driver, since the logger also uses the the same UART port (UART0) the defaults may be conflicting with SCL (the default RTS for UART0)  as documented in issue #10.  This issue will probably come up again with developing the SPI drivers for the LCD and SDCard with the CTS.  It would be best to work on adjusting the logger to use GPIO2 for CTS and GPIO35 for RTS to address and prevent these possible conflicts.

gawlikaj, 2021-05-12:
Currently checking out the data from Waveforms by looking at the Logic Analyzer Traces

gawlikaj, 2021-05-12:
The data from Waveforms indicates that there is something peculiar.  I recognize the start, stop and nack conditions, however I am seeing errors detected on the logic analyzer for the data streamed...

I also noted that there seems to be a way to disable hardware control of the UART (assuming it is not already disabled) which could eliminate such a possible conflict.

gawlikaj, 2021-05-13:
The data from Waveforms indicates that there is something peculiar.  I recognize the start, stop and nack conditions, however I am seeing errors detected on the logic analyzer for the data streamed...

I also noted that there seems to be a way to disable hardware control of the UART (assuming it is not already disabled) which could eliminate such a possible conflict.

For now I will table the Qwiic drivers until after I get to the same point with the SPI drivers.

AJGEmbedded, 2021-08-17:
There seems to be some issues implementing this with the ESP-IDF for now, so a functional test of the existing SparkFun arduino library has been implemented in the way that I intend to use this.  Implementing the project in C/C++ will be revisited later once the rest of the project has been implemented  See #39

AJGEmbedded, 2022-04-01:
After reviewing the arduino libraries, I uncovered that it is necessary to set a timeout value for the I2C bus as of now, the interactive test is working and the component is configurable for instruction timeouts and specific bus speeds.

closed 2022-04-01 by AJGEmbedded