gawlikaj/elek-labirynt #13: Add LiPo "Fuel Gauge" to BOM and hardware

gawlikaj, 2021-04-21:
Add a LiPo fuel gauge to detect the charge level of the rechargeable battery used to power this device.

One that is available that uses the Qwiic connect system (and thus no additional individual wires/traces):
https://www.sparkfun.com/products/10617

gawlikaj, 2021-04-21:
This will be added to the complete board and the controller board after they are designed without it

gawlikaj, 2021-04-22:
Since both the controller and "complete" boards have been designed, I need to do some investigation regarding the LiPo fuel gauge pin layouts and usage.

gawlikaj, 2021-04-22:
Also need to verify that the FreeCad assembly drawings are accurate in terms of pin headers and holes for the boards as per issue #16  so that it is confirmed that the board is designed correctly.  If they need to be redesigned, it could make the design issues more complicated to deal with.

gawlikaj, 2021-04-26:
Since the pin headers and holes for the boards are updated, I am now investigating the LiPo Fuel Gauge.  The datasheet is listed here:  http://cdn.sparkfun.com/datasheets/Prototyping/MAX17043-MAX17044.pdf

In addition to the datasheet, I will also be reviewing the schematics for both the ESP32 Thing Plus U.FL and the LiPo fuel gauge which I have already downloaded in Eagle Format.

gawlikaj, 2021-04-26:
From the schematics, it appears that Vbatt on the ESP32 Thing Plus U.FL would connect to the + signal on the fuel gauge grounding all as appropriate.  The SDA and SCL pins are self explanatory though I may still attempt to use the Qwiic Cable instead since a Qwiic port is provided (which would reduce the number of traces used, however, since I want to ensure that this is going to remain stable, I will most likely use all pins (since there are no drill holes).  This will mount on the bottom side of the board (as indicated on the wiki, bottom mounted peripherals will be on the controller board in the split/2-board version).

gawlikaj, 2021-04-26:
Based off of the reading from the schematics and datasheet the following pin connections should be used:
SDA -> 23
SCL -> 22
QST -> GND (not using hardware Quick Start, will write to the register instead if necessary)
ALT ->  pick one of the unused GPIO pins on the microcontroller:  A2,A3, A5  to incorporate alert interrupt feature, otherwise, do not connect
+    -> Vbatt
-    -> GND

Notes:  my previous schematics suggest to not connect to pins 22 and 23 on the microcontroller as they are also connected to the Qwiic port on the ESP32 Thing Plus, however, since the Qwiic port uses I2C just like the LiPo Fuel Gauge, this should not be a problem, however it will be necessary to test/verify.

AJGEmbedded, 2022-04-02:
So I am working with two different fuel gauges now:  the Qwiic Fuel Gauge - MAX17048 and the SparkFun LiPo Fuel Gauge

The MAX17048 uses the qwiic cable, the jst battery connector, and seemingly would connect to the SparkFun Thing Plus through the battery

The first one actually requires pins and soldering but no holes for standoffs, so ideally I would like to get this working with the second fuel gauge.  Since the driver seems to be working with the second fuel gauge, I will continue down that path for now.  Will need to obtain the appropriate wire, test and crimp.