gawlikaj/elek-labirynt #33: Speaker Circuitry

gawlikaj, 2021-05-13:
Look into adding amplification circuitry between the DAC output and the speakers.

A starting point is here:
https://www.ti.com/lit/an/slyt342/slyt342.pdf

gawlikaj, 2021-05-14:
This should be doable with one op-amp per speaker and one potentiometer per op-amp.  Though this will also require adding some resistors in a voltage divider to get a middle voltage to use for reference.  It would not be a true ground, but it should allow the op-amps to amplify the sound if necessary.

AJGEmbedded, 2021-09-03:
Tested out filter stages separately to filter out DC and high frequencies contributing to quantization error.  Calculating the appropriate resistor values in order to achieve the maximum amplitude allowed by the speakers given that their specifications allow for 0.5W each and the speakers have an impedance of 8 ohms each.  This results in a maximum amplitude of 2V from reference ground.  The maximum gain allowed is thus 1.21 (or 1.6dB)

AJGEmbedded, 2021-09-14:
The speakers still need to be connected to verify that sound will be produced, however the base circuitry has been designed, implemented on a breadboard and tested with the Digilent Analog Discovery 2 oscilloscope feature and waveform captures have been saved.