gawlikaj/elek-labirynt #36: Write Drivers for Momentary Pushbuttons (Buttons 0-4)

gawlikaj, 2021-06-11:
Write the Interface Drivers for the momentary pushbuttons using the GPIO.
This corresponds to buttons 0-4 (red, yellow, green, blue, black respectively)

gawlikaj, 2021-06-11:
Button 0 (Red) corresponds to GPIO32
Button 1 (Yellow) corresponds to GPIO33
Button 2 (Green) corresponds to GPIO12
Button 3 (Blue) corresponds to GPIO13
Button 4 (Black) corresponds to A4 (GPIO36)

gawlikaj, 2021-06-14:
Wired up the circuit and before starting to write the drivers, I decided to check the pins using the logic analyzer from Waveforms... noted that the Red and Yellow buttons appeared to have some random fluctuations and the green button was behaving normally... the blue and black buttons were not responding appropriately.  Taking time to verify the wiring.

When I get to writing the drivers, originally I was planning to use the built-in GPIO settings, however, it seemed like it might be good to use the registers for the purposes of practicing.

This means setting the IO registers for the corresponding GPIO so that they are configured as inputs.  Since I am using 10k ohm resistors, I should not need to use internal pullup resistors.

The random behavior observed for the red and yellow buttons may be due to pre-existing configurations on the ESP32, so for now I will test without having the GPIO connected.

gawlikaj, 2021-06-14:
Reseating the momentary pushbuttons in the breadboard as they did seem a bit loose..  now they are secure in the breadboard and don't jump when pressing/releasing

gawlikaj, 2021-06-14:
Also noted that one half of the breadboard may not transmit the power/ground lines all the way across on it's side ...columns numbered < 30 appear to not be connected to columns numbered > 30 on either side of the breadboard.... so I am keeping the buttons all on the same half of the breadboard and wiring the power supply lines on that side of the breadboard... the microcontroller is remaining on the other half of the breadboard.  Once this has been verified by the Logic Analyzer a picture will be uploaded to the wiki page reflecting the wiring.

gawlikaj, 2021-06-14:
the signals are displaying correctly now

gawlikaj, 2021-06-15:
started the work on the configuration registers and reviewed the technical reference manual regarding their usage for configuring the GPIO buttons and corresponding IO_MATRIX signals

gawlikaj, 2021-06-16:
With a test not using interrupts, button0, button2, and button3 behaved correctly.

I also noticed that when pressing button3 that it also activates an LED on the microcontroller board... would need to investigate what is causing the LED to light up when this should be an output only.

Also button1 and button4 are not responding, which indicates that more investigation is required for them.

gawlikaj, 2021-06-16:
This test was performed while the logic analyzer has been connected to the buttons to confirm correct voltages.

gawlikaj, 2021-06-16:
Upon further inspection when reading the data from the input registers it included extraneous bits.  I have masked those bits out.

Currently only button1 is having a problem with being read correctly.

gawlikaj, 2021-06-16:
After outputting the GPIO_IN_REG and GPIO_IN1_REG, there appears to be no change with regards to pressing the yellow button (Button1).  Next thought is to check to see if there is a problem with the pin for GPIO33

gawlikaj, 2021-06-17:
Using the continuity tester between the button signal terminal and the header pin showed it was connected -- verified wiring
Using the continuity tester between the button signal terminal and the GPIO pin on the ESP32 verified that the signal is connected to where it should be and that there is no connection problem between the ESP32 and the GPIO header pin

On the Technical Reference Manual on page 47 it states that the GPIO Matrix does not need to be configured to read any GPIO pin, but the IO_MUX must be configured to set the FUN_IE bit.  This has already been implemented in the driver.  I may check the forums for information on what is causing GPIO33 to not read the button1 signal.

gawlikaj, 2021-06-17:
Alternatively, I can create a minimal UART driver and remove the existing log reference in the event that the built-in log is causing a problem

gawlikaj, 2021-06-17:
As for the on-board LED, after reviewing the EagleCAD Schematic for the SparkFun Thing Plus ESP32 U.FL, I realized that it is directly connected to GPIO13, so I am not worrying about that at present, especially since GPIO13 is not having an issue.

gawlikaj, 2021-06-17:
From reviewing a discussion on the topic about GPIO33 having difficulty, there is a potential issue related to the XTAL32K register... after reviewing where the RTC IO pins are assigned, they map GPIO33 and GPIO32 to 32K_XN and 32K_XP

gawlikaj, 2021-06-17:
So the 32K_XN and 32K_XP pads need to be routed to the IO_MUX and it has been confirmed by testing to see what happens if the RTCIO_XTAL_32K_PAD_REG bits 17 and 18 were explicitly set as opposed to cleared and button0 failed to respond.  I have reverted to clear.

gawlikaj, 2021-06-17:
While testing things out, the internal pull-down resistors work quite well with the setup

gawlikaj, 2021-06-17:
When using the internal pull-down resistors instead of the external resistors with the setup and reviewing the logic analyzer traces I observed the following:

the Red (button0)   has a lot of noise on it and creates interference with button1, but it displays the correct result on the test application
the Yellow (button1) is generating the correct signal, but has interference with button0 and the test application does not show the output
the Green (button2) is generating the correct signal and the correct result is shown on the test application, but has interference with button1
the Blue (button3) is generating the correct signal and the correct result is shown on the test application, but has interference with button1
the Black (button4) is generating the correct signal and the correct result is shown on the test application

It appears that external pull-downs are better for this case due to signal integrity.  Also taking note that the mystery about GPIO33 is still being investigated.

AJGEmbedded, 2021-06-18:
While trying to figure out the issue with GPIO33, I sent in some questions about it...  in the mean time I started testing with interrupts on the API for a base test... they seem to be working fine (again except GPIO33).  Though I did notice when counting button presses that each press was not a single hit.  Since this appears to be a problem requiring debouncing code I found a resource on debouncing buttons with interrupts on the ESP32 on this link:
https://www.switchdoc.com/2018/04/esp32-tutorial-debouncing-a-button-press-using-interrupts/

AJGEmbedded, 2021-07-01:
I loaded up the latest code on the repository, including testing applications (both standalone and using unity).  There is still a problem with GPIO33 not acting as expected.

AJGEmbedded, 2021-07-01:
A software implementation of switch debouncing has been added so that it does not need to be addressed within a user-defined handler assigned to a button.  This is configurable using idf.py menuconfig  Debounce time is measured in milliseconds.  This has been tested successfully with 200ms as the debounce time.  100ms still allowed 2 triggers when there should have been only one trigger, so it might be possible to try checking between 100ms and 200ms to find the smallest debounce time that will work in order to allow for optimal responsiveness.

AJGEmbedded, 2021-07-01:
The alternative to this software implementation would be to either use an RC filter in order to mitigate the signal bouncing or to use flipflops  both of these strategies are hardware solutions to the same problem.

An RC filter implementation would require calibration of the time constant (R*C) to ensure that the voltage level is at least at the minimum threshold value for a logic level 1 when pressed.  Since the SparkFun Thing Plus ESP32 has 3.3V for it's power supply and for logic level 1, since the threshold value is 0.75*VDD this should come out to 2.475V.  Some circuit analysis will need to be performed in order to ensure that the circuit used will filter as expected.  Since this is still under development, it might be worth exploring and explaining in the wiki.

AJGEmbedded, 2021-07-07:
The alternative implementations for debouncing will be considered later, however to address the issue with the yellow button, I have switched the GPIO pin used for button1 from GPIO33 to GPIO4 (another that was unused.

AJGEmbedded, 2021-07-07:
The driver successfully reads button presses and responds to interrupts, however, a new issue has arose. When pressing a button for a longer period of time the button registers two presses.  Considering a different implementation of debouncing.

AJGEmbedded, 2021-07-07:
Also discovered that pressing buttons rapidly in succession does not register all button presses, this is probably due to the debounce time set being too high.

AJGEmbedded, 2021-07-07:
Since I know that the logic analyzer is recording the signal that I would expect, it seems like I may need to the use the oscilloscope with some of the buttons to identify a more appropriate debounce time.

AJGEmbedded, 2021-08-04:
Drivers have been altered so that debouncing is not in the software.  The software debouncing has been replaced by hardware debounce circuitry.

AJGEmbedded, 2021-08-04:
Further clarification that the circuit used here is using debouncing circuitry with a pull-up resistor and an inverting schmitt trigger instead of a pull-down resistor.  The pull-up resistor and inverting schmitt trigger essentially create the same sort of behavior.  But that discussion is in the #37 presentation and #38 designs.